atc1441/E-Paper_Pricetags #3: Any information on Testpads on the ZBD 50c?

xsrf, 2021-03-27:
Hey, do you have any information on the Testpads of the ZBD 50c? You did connect them to a programmer (?) in https://www.youtube.com/watch?v=OzS4UP7LHaE
I've mapped them but I haven't probed them...
Any useful signals while the display is wake/in sync? I assume you can use them for programming the CC1110?

[image]

xsrf, 2021-03-27:
From the datasheet: 
> The debug interface uses the I/O pins P2_1 as Debug Data and P2_2 as Debug Clock during Debug mode.
> The debug interface allows programming of the on-chip flash. It also provides access to memory and registers contents, and debug features such as breakpoints, single-stepping, and register modification.
> Debug mode is entered by forcing two rising edge transitions on pin P2_2 (Debug Clock) while the RESET_N input is held low.
So it's nice to see these broken out

atc1441, 2021-03-27:
Yes these test pins are used to flash the CC1110, they also have the UART port brought out but until now it is not known what protocol it uses, i speculate on it being the RF data just directly via uart to set its function in production.

on all the ZBD and DisplayData displays these pins are available but not on all in such an order and over the whole pcb

Here is an not so nicely drawn overview
[image]
[image]
[image]
[image]