atc1441/E-Paper_Pricetags #6: ‘Recover display’ seems to use serial of ‘Activate a new display’

jruys, 2021-03-28:
The ‘Recover display’ button seems to use the serial of ‘Activate a new display’, not the one to the left of the ‘Recover’ button.

Both the text in the status field and serial debug show the ‘Activate’ serial #.

atc1441, 2021-03-28:
oh ok thanks for that.

Actual the recover function is not fully tested so its not proven to work but will fix the type right now

xsrf, 2021-03-28:
What does it actually do? From what state does it recover? 🤔

atc1441, 2021-03-28:
It will send a wakeup/activation message with cmd 0x02 https://github.com/atc1441/E-Paper_Pricetags/blob/main/Custom_PriceTag_AccesPoint/ESP32_Async_PlatformIO/RFV3/mode_wu_reset.h#L26

i saw this in the stock firmware named Reset, i hope it resets to default settings, so recover but not shure at all.

it could also just reset the display, but the button is meant for future use with a tested recover method, as the stock software has a recover function, i was not able to trigger it until now

jruys, 2021-03-29:
Tested today’s version - fix comfirmed.

closed 2021-03-29 by jruys