atc1441/E-Paper_Pricetags #7: Boot loop crash after trying to send a bmp

fearthis4, 2021-03-30:
After activation, when trying to send a 1 bit 640x384 monochrome bmp file to the chroma 74 tag, 
From time to time, I get the following repeating in a loop ~10 times:
`input bitmap size: 640 x 384
158868 Main: Count: 4
159082 Main: Mode changed to Sync
159188 Main: Sync
Sync: 0x01 0xfe 0x00 0x00 0x00 0x00

159849 Main: Count: 2
159905 Main: something wrong, back to idle
160059 Main: Mode changed to Idle
160124 Main: Idle
161131 Main: Count: 0
161645 Main: Mode changed to Sync
161819 Main: Sync
Sync: 0x01 0xfe 0x02 0x00 0x00 0x00

162750 Main: Count: 0
162750 Main: something wrong, back to idle
162750 Main: Mode changed to Idle
162750 Main: Idle
163132 Main: Count: 1
163132 Main: Mode changed to Sync
163132 Main: Sync
Sync: 0x01 0xfe 0x00 0x00 0x00 0x00

163150 Normal: short SYNC done
163150 Main: Mode changed to Idle
163151 Main: Idle
E (163818) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (163818) task_wdt:  - async_tcp (CPU 1)
E (163818) task_wdt: Tasks currently running:
E (163818) task_wdt: CPU 0: IDLE0
E (163818) task_wdt: CPU 1: loopTask
E (163818) task_wdt: Aborting.
abort() was called at PC 0x40147bf8 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40088c50:0x3ffbf9a0 0x40088ecd:0x3ffbf9c0 0x40147bf8:0x3ffbf9e0 0x40087405:0x3ffbfa00 0x4015a5af:0x3ffbc330 0x4014a30b:0x3ffbc350 0x4008b68d:0x3ffbc370 0x40089ede:0x3ffbc390

Rebooting...
`
If this happens I have to reactivate the display to successfully send a cmd again with the display tag.

atc1441, 2021-03-31:
Could you please upload the image if possible. 

It looks to me like the image is not being compressed as it has to much dithering so the compression takes longer than the next slot where the image is already tried to send to the display. 
Just a guess

fearthis4, 2021-03-31:
Attached are the files.
I took the files and just created them in paint at the correct size then ran them through imagemagick:
convert <input_image> -depth 1 -type bilevel BMP3:<output_image>.bmp
[images.zip](https://github.com/atc1441/E-Paper_Pricetags/files/6234459/images.zip)

atc1441, 2021-03-31:
The format i use is monochrome windows bmp
You can save the image with paint directly as monochrome bmp

Will try it this evening, sounds like the bmp header is wrong. Can not check right now.

The image overall should work

fearthis4, 2021-03-31:
[image]
I got the out1.bmp to work with a couple tries, but after an erase it did not want to do the second with the out2.bmp in the second image slot.

atc1441, 2021-03-31:
[image]

Was able to upload it on first try but the next one failed, it does indeed looks like the compression is still ongoing and already send to the display so it crashes because both functions try to acces the data. the compression is done in another thread then the RF part. 

Looks like it needs a flag to check if the compression is already done before starting to send the data

fearthis4, 2021-04-01:
Nice, good catch.
Thanks for all of the help Aaron.

Hades32, 2021-04-02:
This happened to me as well when trying to send the provided demo files (75testC2.bmp and 75test.bmp). Only a single file didn't produce the crash

atc1441, 2021-04-02:
Just looked into this, but it seems like the flag for new data is only set after the compression is done already. https://github.com/atc1441/E-Paper_Pricetags/blob/main/Custom_PriceTag_AccesPoint/ESP32_Async_PlatformIO/RFV3/web.cpp#L102

so it may has another cause. not shure what right now

atc1441, 2021-04-02:
just published an update which seem to fix the issue, 
setting also the id of the displays only after the image is compressed not only makes sense but seem to help. 
didn't had any crash after changing it

fearthis4, 2021-04-29:
This is no longer an issue in the latest version tested from April 18th 2021..

closed 2021-04-29 by fearthis4