atc1441/E-Paper_Pricetags #8: Stuck Chroma29 - suggestions?

jruys, 2021-03-31:
I have a Chroma29 that used to work for me (I managed to load a .bmp at some point) but it seems stuck now.

 - Played with Freq offset
 - Tried activating on different Freq
 - Removed batteries for a while and replaced with fresh
 - Tried "recover" (not sure if that should help)
 
 Any thoughts what to try/debug next? I have three others that work fine, so my ESP32/CC1101 setup seem ok.

atc1441, 2021-03-31:
Hey, the Chroma29 is still the one with the most problems currently. so it is not fully sure how to recover it. 
but working on it these days as it bugs me too much

The recover button sends a wakeup request with cmd 2 which is called in the stock software "reset" but its not sure what it means

atc1441, 2021-03-31:
just made an update with  a fix for the activation of the Chroma29

the problem was that the select slot was wrong, on the old activation method the slot is selected automatically to the LSBs but on the  new one it needs to be selected on activation which was not did right.

I can now finally control the ones i activated. this was correct some time ago but on some sorting i seem to have added that bug...

jruys, 2021-03-31:
Fix works! It’s talking to me again...

closed 2021-04-01 by jruys