atc1441/E-Paper_Pricetags #15: Chroma29 unpredictable behaviour

jruys, 2021-04-10:
I'm having unpredictable results with the Chroma29. Some activate fine but won't work the next day, others don't seem to activate but respond to commands now and again. They will swap behaviour, one that works fine now will misbehave tomorrow.

Any known tricks to make them behave? Using red CC1101, frequency tuned with Airspy and powering from lab PSU.

Recent answers.txt:
NetID: 1 freq: 2 display: 0 87184 = 85:05:19:0C:CA:00:01:00:00:00
NetID: 254 freq: 8 display: 0 195477 = 12:9F:09:2F:00:00:03:00:00:04:00:00:9A:11:80:00:28:01:01:01:06:00:00:02:08:00:01:02:21:01:00:21:22:24:00
NetID: 254 freq: 8 display: 0 59080 = 12:9F:09:2F:00:00:03:00:00:04:00:00:9A:11:80:00:28:01:01:01:06:00:00:02:08:00:01:02:21:01:00:21:22:24:00
NetID: 1 freq: 2 display: 0 303888 = 85:05:00:00:00:00:06:00:00:00
NetID: 1 freq: 2 display: 0 321698 = 03:04:85:05:AD:0B:E8:00:01:00
NetID: 1 freq: 2 display: 0 343918 = 03:04:85:05:9F:0B:E8:00:01:00
NetID: 1 freq: 2 display: 0 629488 = 85:05:B8:0B:CE:00:01:00:00:00
NetID: 254 freq: 8 display: 0 678380 = 12:9F:09:2F:00:00:03:00:00:04:00:00:9A:11:80:00:28:01:01:01:06:00:00:02:08:00:01:02:21:01:00:21:22:24:00
NetID: 1 freq: 2 display: 0 155388 = 85:05:00:00:00:00:06:00:00:00
NetID: 1 freq: 2 display: 0 177598 = 03:04:85:05:DB:0B:AA:00:01:00
NetID: 1 freq: 2 display: 0 323018 = 03:04:85:05:D2:0B:AF:00:01:00
NetID: 254 freq: 8 display: 0 402294 = 12:9F:09:2F:00:00:03:00:00:04:00:00:9A:11:80:00:28:01:01:01:06:00:00:02:08:00:01:02:21:01:00:21:22:24:00
NetID: 1 freq: 2 display: 0 837818 = 03:04:85:05:BF:0A:C2:00:01:00
NetID: 1 freq: 2 display: 0 275290 = 85:05:24:0B:DB:00:01:00:00:00

atc1441, 2021-04-10:
Hey, 

it looks like you are using the same id for the display? or is this just the log of that one ?

try to use a different id each time you activate one just for testing please

jruys, 2021-04-10:
I am using different IDs (124..127) but somehow the log always shows “display: 0”. In the past it showed the assigned ID there but that changed a few builds ago?

atc1441, 2021-04-10:
just looked after that but haven't found the cause, but there must indeed be some bug then, but should not be related to the issue here

xsrf, 2021-04-18:
@jruys I just pushed a fix that made the activation of the Chroma29 way more reliable for me. Maybe it helps you too.
If you're not sure if it is already activated, force it to lose sync by switching your AP off (or to a different channel) for ~2min. Then switch back on / to your desired channel and run the activation. The activation is now followed by a full sync and then it should be ready for communication.

probonopd, 2021-04-19:
> I am using different IDs (124..127) but somehow the log always shows “display: 0”. In the past it showed the assigned ID there but that changed a few builds ago?

This is fixed now.