atc1441/E-Paper_Pricetags #17: ESP32 Access Point wiring

probonopd, 2021-04-11:
> Connect the CC1101 868 MHz radio module to the ESP32 using the pinout specified in the RFV3.h header file.

https://github.com/atc1441/E-Paper_Pricetags/blob/78a44580edc0f980931f94b63ca98ce65c26ddf2/Custom_PriceTag_AccesPoint/ESP32_Async_PlatformIO/RFV3/RFV3.h#L22-L28

Does this mean one should connect the yellow and the grey wire **both to the same pin** on the ESP32?

[image]

[Graphic source](https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/master/img/Wiring_CC1101.png)

atc1441, 2021-04-11:
Ah sorry, the GDO0 pin is currently not used, so you can leave it unconnected. 

it will maybe be used in the future but not for now

probonopd, 2021-04-11:
Thank you.
Will watch **https://www.youtube.com/watch?v=pRoFim3Egss**, probably everything is explained there :+1:

closed 2021-04-11 by probonopd