atc1441/E-Paper_Pricetags #20: Do not hardcode the WLAN credentials

probonopd, 2021-04-13:
Let's not hardcode the WLAN credentials. using WiFiManager the device opens a temporary WLAN access point that can be used to tell the device which WLAN to connect to, and using which password.

[image]

[Screenshot source}(http://www.martyncurrey.com/esp8266-and-the-arduino-ide-part-5-adding-wifimanager/)

I have started to integrate this but since I am still waiting on devices I could not test this on hardware yet:

https://github.com/atc1441/E-Paper_Pricetags/compare/main...probonopd:patch-1

atc1441, 2021-04-13:
Used it before with other projects and quite satisfied with it.

For simplicity i left hard coded credentials in development of the RF protocol in but got no problem with implementing it, also the usual OTA is still needed to be implemented :)

probonopd, 2021-04-16:
Closed by https://github.com/atc1441/E-Paper_Pricetags/pull/21

closed 2021-04-16 by probonopd