atc1441/E-Paper_Pricetags #23: Clarify save_settings

probonopd, 2021-04-16:
If one uses "Set base freq offset" and clicks on "save_settings", is this setting then automatically applied after a reboot of the ESP32 or does one need to click "get settings" in order to load it? I am asking because after a reboot the web page does not seem to have "remembered" the settings.

Also, does one have to cleick "Do sync" after a reboot or does it do the sync (using the settings) automatically=

atc1441, 2021-04-16:
The offset is saved  with save_settings and will be used on reboot.

The get settings button will get the current settings, it is still missing to parse that answer and set it in the input fields to "remember" it.
So on load of the .htm a one time request needs to be implemented that will request the current settings.

A sync is done every 500 periods, so on reboot it needs to be fired manualy to get a quick sync

kgarrels, 2021-04-16:
could we foresee an option to do a sync after reboot?

atc1441, 2021-04-16:
It was there but i removed it for development. Its simple to add again :)

kgarrels, 2021-04-16:
ups, sorry…

kai

atc1441, 2021-04-16:
You can set this value 
https://github.com/atc1441/E-Paper_Pricetags/blob/9d5d92b2e360d194a858258c493155ced9ee8d33/Custom_PriceTag_AccesPoint/ESP32_Async_PlatformIO/RFV3/mode_idle.h#L45

To 490 and it will resync on reboot :)

probonopd, 2021-04-16:
Is there a reason why we don't do this by default?

atc1441, 2021-04-16:
Because if an activation is wanted to perform directly after reboot it can get problematic.

The displays already received an wakeup cmd then and will not listen to the activation cmd anymore.

Thats what i meant i removed it for development, it is "activated" again and the esp32 is running for a while it makes no difference anymore