atc1441/E-Paper_Pricetags #24: Build using native Arduino toolkit fails

jruys, 2021-04-16:
Haven't had time to dive into this, opening just in case others run into this. Build using native Arduino toolkit started to fail after WifiManager merge.

Installed the library from https://github.com/tzapu/WiFiManager (not via native library manager)

Build exits with "'class WiFiClass' has no member named 'mode'"
Compile errors start with "error: redeclaration of 'HTTP_DELETE'   XX(0,  DELETE,      DELETE)"

Does WifiManager have version dependencies I may not have taken into consideration?

atc1441, 2021-04-16:
Maybe this helps ? 
https://github.com/atc1441/E-Paper_Pricetags/blob/main/Custom_PriceTag_AccesPoint/ESP32_Async_PlatformIO/platformio.ini#L19

havent tested it myself in Arduino

probonopd, 2021-04-17:
Yes, you definitely need to use the `feature_asyncwebserver` branch of https://github.com/tzapu/WiFiManager: https://github.com/tzapu/WiFiManager/archive/refs/heads/feature_asyncwebserver.zip

jruys, 2021-04-17:
Confirmed: switching to that library fixes the issue. It now compiles ok.

probonopd, 2021-04-17:
Sent a pull request to clarify the instructions. Thanks for pointing this out.

closed 2021-04-17 by atc1441