atc1441/E-Paper_Pricetags #27: How to read voltage and temperature from Chroma29

probonopd, 2021-04-17:
According to https://github.com/atc1441/E-Paper_Pricetags/blob/caaacc2b2d376a3cea1ab35703f6107dd4a83df9/Custom_PriceTag_AccesPoint/CMDs%20explanation.txt#L129-L132
the answer to command `85` contains the voltage and temperature.

Sending command `85` makes the whole display black. And we see something like this:

```
last answer: Msg from display NetID: 1 freq: 0 display: 0 8044533 = 85:05:71:0B:EC:00:01:00:00:00
```

How to interpret the answer?

* `85:` The command that this is the answer to
* `05:` Length of this answer is 5 bytes
* `71:0B:` Hex 0x0B71 = Decimal 2929 = 2.929 Volts
* `EC:00:` Hex 0x00EC = Decimal 236 = 23.6 degrees Celsius
* `01:` Unknown?

So, how can we get the values without making the display black?

According to the same page, command `02` should give us the voltage and command `19` should give us the temperature.

Just sending the command `02` results in

```
last answer: Msg from display NetID: 1 freq: 0 display: 0 8365754 = ��?
```

Why is there garbled output instead of the hex answer?

Can we/do we need to combine multiple commands, like sending command `0219` for getting both the voltage and the temperature? This gives

```
last answer: Msg from display NetID: 1 freq: 0 display: 0 9135793 = 82:02:F5:00:99:03:7A:0B:00:00
```

How to interpret the answer?

atc1441, 2021-04-17:
Also after an image is send the temp and battery is send back just as info.

The garbage answer you got back at 02 is an error, it should return something like you got int the 0219 cmd. 82:02:F5:00

The 0219 cmd can be parsed the following:
82 = cmd 2 with incoming data
02 = 2bytes of data incoming
F5 = actual data byte 1
00 = actual data byte 2

so 0x00f5 = 24.5 degrees Celsius

99 = cmd 10 with incoming data
03 = 3bytes of data incoming
7A = actual data byte 1
0B = actual data byte 2
00 = actual data byte 3

0x0b7a = 2.938 Volts

00 = the length of replying data is always is a pack of 5bytes so this 00 is not used 

as you can see multible cmds can be send at once, when sending an image update 3 cmds are send at once, header of img, actual img and the refresh cmd itself. the refresh cmd always return the temp and battery data in return

probonopd, 2021-04-18:
Maybe we should parse that data and output it whenever we receive it?

atc1441, 2021-04-18:
Yes.
A dynamic incoming cmd parser is still needed to be written, as also the display answers with multiböe cmds in one go, on display update you get answer 03 04 85 ...
But on resfresh only 85 ...

So that is needed to be parsed correctly, also the incoming ack needs to be looked at, as if a msg is send again it will currently be added to the answer buffer even if its the same as already received.

I did not do anything with the answer buffer for now as i concentrated on the system itself