atc1441/E-Paper_Pricetags #29: Always says "display: 0"

probonopd, 2021-04-17:
Always says "display: 0" in `answers.txt` and in the Status area.

https://github.com/atc1441/E-Paper_Pricetags/blob/e9588c02beff5b0430add6320040759919cb6710/Custom_PriceTag_AccesPoint/ESP32_Async_PlatformIO/RFV3/trans_assist.cpp#L118

probonopd, 2021-04-17:
Looks like `set_display_id(id);` is _only_ ever called in `server.on("/set_custom", HTTP_POST, [](AsyncWebServerRequest *request)`, is that correct?

atc1441, 2021-04-17:
The display id is used to send to the correct displays so is set correctly, my guess is that it is cleared to early so back 0

atc1441, 2021-04-17:
The problem is here https://github.com/atc1441/E-Paper_Pricetags/blob/e9588c02beff5b0430add6320040759919cb6710/Custom_PriceTag_AccesPoint/ESP32_Async_PlatformIO/RFV3/main_variables.cpp#L102

atc1441, 2021-04-17:
If we put save receive buffer above this one it will work again https://github.com/atc1441/E-Paper_Pricetags/blob/e9588c02beff5b0430add6320040759919cb6710/Custom_PriceTag_AccesPoint/ESP32_Async_PlatformIO/RFV3/mode_trans.h#L25

PR #32 (2021-04-18): Fix Display ID in responses
`save_receive_buffer();` earlier, closes #29
closed 2021-04-18 by atc1441