atc1441/E-Paper_Pricetags #30: Idea: MQTT publish/subscribe

probonopd, 2021-04-17:
It would be nice if one could use [MQTT](https://mqtt.org/) and [Node-RED](https://nodered.org/) to send messages out to the displays, and get messages from the displays.

This way, the displays could be hooked much more easily e.g., into home automation setups.

The following is not quite working yet because I am not sure how to hand over the values properly, but should give a rough idea.

https://github.com/atc1441/E-Paper_Pricetags/compare/main...probonopd:codespace-06f9

What do you think, would this be interesting and worthwhile for you?

atc1441, 2021-04-17:
Sound like an interesting idea and is possible.

The dynamic text is currently handled in web.c i could either also be done their or the text parsing needs an own file in general @bitbank2 added this feature.

I do not use mqtt at home but it should be quite simple to get an test setup runing to get it working

probonopd, 2021-04-17:
Actually I am most interested in getting the messages (responses) from the displays published on MQTT so that e.g., Nore-RED can watch the battery levels and temperatures for each display. Sending data to the devices can for now be done using HTTP requests.