atc1441/E-Paper_Pricetags #33: Garbled responses for some commands

probonopd, 2021-04-18:
Responses for some commands are garbled, for example:

```
OK cmd to display 243 0000 Len: 2
Send: successful , waiting: 0
Activation: not started
NetID 1 freq 0 slot 2 bytes left: 0 Open: 0
last answer: Msg from display NetID: 1 freq: 0 display: 243 139954 = �����?��??
mode Idle
```

atc1441, 2021-04-18:
In this one you send 

00 00 as cmd right?

That is no valid cmd. So the answer from the display is nothing. When nothing is replied the uninitialized answer buffer will be used. By adding a filter for 0 len answers it should be fixed

atc1441, 2021-04-18:
Something like this 
https://github.com/atc1441/E-Paper_Pricetags/blob/main/Custom_PriceTag_AccesPoint/ESP32_Async_PlatformIO/RFV3/trans_assist.cpp#L122

data_received += length_to_receive?str:"No answer";

Not tested

probonopd, 2021-04-18:
When I do this, then it looks like it somehow breaks things, commands don't get properly sent and responses get timeouts.

```
PacketCounter: 8 rx: 0Len 16 Data: 0x01 0x00 0x2d 0x07 0x00 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff
68269 : RX_TIMEOUT!!!
68269 : TIME LEFT: 102
68269 : Mode changed to Idle
68269 : Idle
```

atc1441, 2021-04-18:
That looks more like a real timeout? The change is not called before a full received msg. But just guessing here

probonopd, 2021-04-18:
Maybe I am running into intermittent errors due to low-ish batteries, could it be? 
My theory is that after changing the display content, the batteries may need some minutes to "recover"?