atc1441/E-Paper_Pricetags #34: Crash when sending default text

probonopd, 2021-04-18:
Trying to send the default text to a Chroma29:

```
Enter text here. Each display type uses a font which fits about 18 characters per line. Press the 'Send Text' button above to transmit it.
```

leads to a crash which reboots the ESP and puts it into WiFiManager captive portal mode:

```
133289 : short SYNC done
133289 : Mode changed to Idle
133289 : Idle
Entering set_custom
Display type = 2

Stack smashing protect failure!

abort() was called at PC 0x401607f3 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x40089538:0x3ffe7f00 0x400897b5:0x3ffe7f20 0x401607f3:0x3ffe7f40 0x400d8595:0x3ffe7f60 0x400d8dd9:0x3ffe8030 0x400d91a5:0x3ffe8190 0x400e1131:0x3ffe81b0 0x400def19:0x3ffe81f0 0x400defa5:0x3ffe8240 0x401851c1:0x3ffe8260 0x40185251:0x3ffe82a0 0x401858f6:0x3ffe82c0 0x4008a7c6:0x3ffe82f0

Rebooting...
```

With this text it does not happen:

```
Enter text here. Each display type uses a font which fits about 18 characters per line.
```

Maybe some memory is too small?

probonopd, 2021-04-18:
120 characters = works
130 characters = crash

atc1441, 2021-04-18:
Can confirm the Behavior, unfortunately do not have a quick solution for that one, could be related to the async webserver or to the custom code itself
Needs to be debuged further

probonopd, 2021-04-18:
I'd say let's at least use another default message for now so that things don't crash when people just click on "Send Text" without changing the default ;-)