atc1441/E-Paper_Pricetags #37: Getting WakeUp during activation / resync counter

xsrf, 2021-04-18:
Hey,
it sometimes switches into WakeUp mode during activation of the Chroma29 ESLs. I guess this is because you're switching to sync/idle mode during activation on the activation channel and `resync_start_counter` may just kick in and mess with the activation, switching to WakeUp.
I guess adding a check in `mode_idle.h` would help 
```c
    if (resync_start_counter >= get_rounds_to_resync() && !get_last_to_short()  && !get_trans_mode())
```
But I'm wondering if it wouldn't be better to put the resync counter into a global scope and make sure it only triggers a resync after it was in sync/idle mode for the required intervals and reset it completely whenever an other mode is active?

atc1441, 2021-04-18:
Yes the counter should be better in main_variables and check for further states before going into wakeup mode

xsrf, 2021-04-18:
Actually, checking additional states wouldn't be required if modes other than sync+idle would just reset the counter.

PR #39 (2021-04-18): Reliable activation and sync with Chroma29
Prevent occasional wu+sync during activation (Fixes #37)

I constantly had issues with the activation of the Chroma29 ESLs. They sync to the activation channel and get their configuration properly but then lost sync. I guess the switching back to the wakeup channel after activation caused this. With just a full sync after activation the problem is gone and I can communicate with the Chroma29s reliable right after activation.
closed 2021-04-18 by atc1441