atc1441/E-Paper_Pricetags #40: Feature Request: Automatically load pictures from URL

Hades32, 2021-04-25:
It would be great if it was possible to configure per activation ID (and per color) a URL [pointing to a BMP](https://raw.githubusercontent.com/atc1441/E-Paper_Pricetags/main/Custom_PriceTag_AccesPoint/demoFiles/chroma74.bmp) and a refresh interval.

The ESP could then grab these images in the defined interval, check if they changed (either via `ETag`, `If-Modified-Since` or checksum) and send them to the device if they changed.

This way it would be easy to host image creation on more powerful devices or in the cloud. We could even build parametrizable services that everybody can use - something like https://epaper.dev/date-news-weather?device=c74&color=b&city=munic,de which could compose images from some fragments and return them in a device-specific format

Hades32, 2021-04-25:
I was wondering if this is even possible or if downloading via HTTPS would be too time-consuming and thus break the tags' protocol timing...

probonopd, 2021-04-25:
One could use the spiffs on the ESP as a cache for the image data.

atc1441, 2021-04-25:
This feature is planed.

A general task handler with also some dynamic content is a good solution for many different requests

Hades32, 2021-04-25:
@atc1441 I think you mentioned wanting to add a database of known activations. Is there a feature branch for this already?
If I find time to work on this PR it would make sense to build it on top of such a database feature

atc1441, 2021-04-25:
No its not started at all.
I tested a few different solutions for databases.

The winner is still a simple txt file which is loaded "manual" and parsed that way.

One other option was sqlite but that is hard to edit without an editor or only via the sql requests

Hades32, 2022-01-16:
FYI, it's not yet fully automatic (in the sense of polling on an interval), but it's working on my fork now:
* the fork https://github.com/Hades32/web-tags
* a cloud solution to create the images: https://github.com/Hades32/eink-bmp-worker