atc1441/E-Paper_Pricetags #41: On latest release 04-18-21, "No connection possible" on Chroma74 activation

fearthis4, 2021-04-30:
When trying to test out the Chroma74 display activation on the latest build it keeps getting RX Timeouts with a no connection possible message:
``2692061 : short SYNC done
2692061 : Mode changed to Idle
2692061 : Idle
2692663 : Mode changed to Wakeup Activation
2692663 : Wakeup Activation
Len 10 Data: 0x00 0x4a 0x4d 0x10 0x17 0x66 0x78 0x01 0x00 0x01
2692674 : Radio set to base freq: (F=35) 915.00 Mhz 
2692674 : Radio set id done 0
2693143 : Count: 109
2694243 : Count: 248
2695343 : Count: 248
2696443 : Count: 248
2697543 : Count: 250
2698643 : Count: 247
2699743 : Count: 247
2700843 : Count: 248
2701943 : Count: 248
2703043 : Count: 248
2704143 : Count: 248
2705243 : Count: 249
2706343 : Count: 248
2707443 : Count: 249
2708167 : Wakeup Activation done
2708167 : Mode changed to Activation
2708167 : Activation
Len 16 Data: 0x00 0x4a 0x4d 0x10 0x17 0x66 0x78 0x80 0x03 0x01 0x00 0x7b 0x00 0x00 0x00 0x00
2708213 : RX Timeout retry
Len 16 Data: 0x00 0x4a 0x4d 0x10 0x17 0x66 0x78 0x80 0x03 0x01 0x00 0x7b 0x00 0x00 0x00 0x00
2708259 : RX Timeout retry
Len 16 Data: 0x00 0x4a 0x4d 0x10 0x17 0x66 0x78 0x80 0x03 0x01 0x00 0x7b 0x00 0x00 0x00 0x00
2708305 : RX Timeout retry
Len 16 Data: 0x00 0x4a 0x4d 0x10 0x17 0x66 0x78 0x80 0x03 0x01 0x00 0x7b 0x00 0x00 0x00 0x00
2708351 : RX Timeout retry
Len 16 Data: 0x00 0x4a 0x4d 0x10 0x17 0x66 0x78 0x80 0x03 0x01 0x00 0x7b 0x00 0x00 0x00 0x00
2708397 : RX Timeout retry
Len 16 Data: 0x00 0x4a 0x4d 0x10 0x17 0x66 0x78 0x80 0x03 0x01 0x00 0x7b 0x00 0x00 0x00 0x00
2708443 : RX Timeout retry
Len 16 Data: 0x00 0x4a 0x4d 0x10 0x17 0x66 0x78 0x80 0x03 0x01 0x00 0x7b 0x00 0x00 0x00 0x00
2708489 : RX Timeout retry
Len 16 Data: 0x00 0x4a 0x4d 0x10 0x17 0x66 0x78 0x80 0x03 0x01 0x00 0x7b 0x00 0x00 0x00 0x00
2708535 : RX Timeout retry
Len 16 Data: 0x00 0x4a 0x4d 0x10 0x17 0x66 0x78 0x80 0x03 0x01 0x00 0x7b 0x00 0x00 0x00 0x00
2708543 : Count: 173
2708581 : RX Timeout retry
Len 16 Data: 0x00 0x4a 0x4d 0x10 0x17 0x66 0x78 0x80 0x03 0x01 0x00 0x7b 0x00 0x00 0x00 0x00
2708627 : RX Timeout retry
Len 16 Data: 0x00 0x4a 0x4d 0x10 0x17 0x66 0x78 0x80 0x03 0x01 0x00 0x7b 0x00 0x00 0x00 0x00
2708674 : no connection possible, exit activation
2708674 : Mode changed to Idle``

Tried with both US and EU channel.

Using the old offset that worked previously with earlier builds (15)
However when trying EPOP900 display it works the first time, although actually sending data is getting RX timeouts.

atc1441, 2021-04-30:
Hey. It is more likely related to the displays.

There was no untested change lately maybe ypu can try new batterys or direct powering for test

fearthis4, 2021-04-30:
Just went through and tested the batteries and everything seems fine, I will try to do some further testing and follow-up.