atc1441/E-Paper_Pricetags #42: Compiling error with ESP32 / Wifi-Manager (Arduino)

DL5SMB, 2021-05-02:
I tried compiling the newest release of the Custom Access Point with Platformio and Arduino without much success.
With Platformio the code compiled and uploaded, but did not do anything (no WIFI Access Point appeared).

On Arduino compiling stopped after logging some warnings, notes and the following error message:

`C:\Users\DL5SMB\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver\WiFiManager.cpp: In member function 'void WiFiManager::WiFi_autoReconnect()':
C:\Users\DL5SMB\Documents\Arduino\libraries\WiFiManager-feature_asyncwebserver\WiFiManager.cpp:3486:79: error: no matching function for call to 'WiFiClass::onEvent(std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type)'
       wm_event_id = WiFi.onEvent(std::bind(&WiFiManager::WiFiEvent,this,_1,_2)); // @todo move, needed for async esp32 scannetworks`

I think this is a bug of the Wifi-Manager library and the ESP32. Maybe someone can verify this?

I checked out the code before the change to this library, and the old code compiled without errors and the ESP32/Webserver
starts and is accesible.

Sascha

atc1441, 2021-05-02:
I had the problem that multible comoilings where needed so maybe try that once.

Will most likely reverd the latest changes for now

DL5SMB, 2021-05-02:
I did try several compiling runs. It aborted every time without success.

The older version seems to work, I was able to at least activate several Chroma74 displays,
though I could not do anything more.
Took some time since the radio module  was off about a value of "35" (checked it with my SDR).

All other buttons do not seem to activate the radio module and if i try to synchonize I get the 
message on the web server "ERROR: TypeError: NetworkError when attempting to fetch resource."
And absoluteley no output on the serial monitor.

If i try to send a text message, I get the serial output
'00:26:33.880 -> Entering set_custom
00:26:33.880 -> Display type = 0
00:26:33.880 -> 83,19,00,5a,00,e0,00,0d,07,00,00,00,00,00,00,00,00,00,00,00,00,00,08,01,00,00,00,84,87,0d, 
00:26:33.880 -> Writing 1842 bytes to temp.bin'
And then it stopps.
And if i try to send the demo chroma74.bmp I get the message
'OK cmd to display 127 File: chroma74.bmp Len: 9538
Send: in sending , waiting: 1
Activation: successful
NetID 2 freq 2 slot 0 bytes left: 0 Open: 0
last answer: No data received so far
mode Idle
⚫'
from the webserver and 
'00:29:14.051 -> input bitmap size: 640 x 384
00:29:14.051 -> Entering encode_raw_image; width=640, height=384
00:29:15.862 -> 83,19,00,80,02,80,01,1b,25,00,00,00,00,00,00,00,00,00,00,00,00,00,4c,02,00,00,00,84,80,00,25,1b, 
' as serial output. And then it stops again

Maybe some fresh batteries will help. And then - just for comparing - I will try to a.) check the new code on an
ESP8266 and b.) try out another transmitter module

Sascha

probonopd, 2021-05-03:
> Maybe some fresh batteries will help. 

Did you press "wakeup and sync" and waited until it said "Full sync"? After that I can activate and send pictures.

> a.) check the new code on an ESP8266 and 

I didn not get it to compile there, probably some spiffs-related changes would be needed.

DL5SMB, 2021-05-03:
Yes, I did - and I tried it several times.
Depending on the browser I also get different error messages around 5 seconds after pushing the "wake + full sync" button.

With Firefox it is 
```
**ERROR: TypeError: NetworkError when attempting to fetch resource.**
Send: nothing send , waiting: 0
Activation: successful
NetID 2 freq 2 slot 0 bytes left: 0 Open: 0
last answer: No data received so far
mode Idle
⚫
```

With Edge or Chrome it is
```
**ERROR: TypeError: Failed to fetch**
Send: nothing send , waiting: 0
Activation: successful
NetID 2 freq 2 slot 0 bytes left: 0 Open: 0
last answer: No data received so far
mode Idle
```

atc1441, 2021-05-04:
It sounds to me that some lib had an update and may did broke something.

We should providecour versions of them to compare

DL5SMB, 2021-05-04:
These are the lib versions the Arduino-IDE prints out during compiling:

```
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: C:\Users\DL5SMB\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Nicht benutzt: E:\Program Files (x86)\Arduino\libraries\WiFi
Bibliothek SPI in Version 1.0 im Ordner: C:\Users\DL5SMB\Documents\Arduino\hardware\espressif\esp32\libraries\SPI  wird verwendet
Bibliothek FS in Version 1.0 im Ordner: C:\Users\DL5SMB\Documents\Arduino\hardware\espressif\esp32\libraries\FS  wird verwendet
Bibliothek SPIFFS in Version 1.0 im Ordner: C:\Users\DL5SMB\Documents\Arduino\hardware\espressif\esp32\libraries\SPIFFS  wird verwendet
Bibliothek ESPmDNS in Version 1.0 im Ordner: C:\Users\DL5SMB\Documents\Arduino\hardware\espressif\esp32\libraries\ESPmDNS  wird verwendet
Bibliothek WiFi in Version 1.0 im Ordner: C:\Users\DL5SMB\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi  wird verwendet
Bibliothek AsyncTCP im Ordner: C:\Users\DL5SMB\Documents\Arduino\libraries\AsyncTCP (legacy) wird verwendet
Bibliothek ESPAsyncWebServer in Version 1.2.3 im Ordner: C:\Users\DL5SMB\Documents\Arduino\libraries\ESPAsyncWebServer  wird verwendet
Bibliothek OneBitDisplay im Ordner: C:\Users\DL5SMB\Documents\Arduino\libraries\OneBitDisplay (legacy) wird verwendet
Bibliothek BitBang_I2C im Ordner: C:\Users\DL5SMB\Documents\Arduino\libraries\BitBang_I2C (legacy) wird verwendet
Bibliothek Wire in Version 1.0.1 im Ordner: C:\Users\DL5SMB\Documents\Arduino\hardware\espressif\esp32\libraries\Wire  wird verwendet
```
Espressif/esp32 is the newest Version from 28.04.21 from the Master branch on Github.
xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win32.zip

atc1441, 2021-05-06:
Sorry for the late reply, i just checked the versions and it seems to fit.

very strange for why you have such problems, can you maybe try a different ESP32 ?

DL5SMB, 2021-05-06:
So, I just reinstalled everything on my second computer on Linux, this time using the release version 1.06 (I think) of the Espressif Arduino IDE. And now  I was able to use the latest version of the E-Paper-Pricetag-Firmware. It seems to work, I just succcesfully uploaded some pictures.

DL5SMB, 2021-05-07:
I just started experimenting with three colours. Is there a recommended way to create the second picture / photo with the additional colour?

Is the third colour (second picture) just overlayed, or do you need to erase the black pixels on the first picture  at the corresponding places?

I tried selecting and copying coloured areas of a a photo with The Gimp and then pasting and saving it as new file before converting both files two BMP pictures. Both pictures easily go out of Sync if done this way.
Then I tried  copying the first picture and deleting areas on the copy which were covered by the colours on the first photo. No results I am afraid.

Also, is there more informatiom on  how Dimitrys Soft and Firmware can be used on the stock displays and the Modul? I would love to see some greyscalebp ictures.

Sascha

atc1441, 2021-05-08:
Hey.

Please try out the /draw.htm https://github.com/atc1441/E-Paper_Pricetags/blob/main/Custom_PriceTag_AccesPoint/ESP32_Async_PlatformIO/data/draw.htm
There you can cry easily see how the colors behave.

Dmitrys firmware is currently only supported on an nRF52840, the ESP32 AP code from here can be used to update to his firmware