atc1441/E-Paper_Pricetags #43: 900MHz version of the Chroma 74

chaseadam, 2021-06-02:
You may wish to note that there is a US version of the Chroma 74 which runs in the 900MHz band. These units will have "FCC ID: VC7120-0157" on the back label.

Specifically they run at 902.5 to 927.5 MHz per [FCC documents](https://apps.fcc.gov/oetcf/eas/reports/ViewExhibitReport.cfm?mode=Exhibits&RequestTimeout=500&calledFromFrame=Y&application_id=wc7HDE0Te%2FgtLU%2BZOKqPsg%3D%3D&fcc_id=VC7120-0157)

It seems the C1101 transmitter recommended will work with this as well. I ordered a few, but haven't tested them yet.

atc1441, 2021-06-02:
Hey, the Displays can do both 868 and 915mhz, 

they will listen on a set activation channel.

the 915mhz is also supported with the ESP32 firmware.

chaseadam, 2021-06-03:
That is great news! All the talk about 868MHz made me think there might be a regional version. Excited to try them out.

Thanks for the reply and all of the details!

closed 2021-06-03 by chaseadam