atc1441/E-Paper_Pricetags #46: Can't get interrupts to work

skiphansen, 2021-11-27:
I'm pretty sure this isn't an issue with Aaron's code, but I don't know where else to ask.

I'm not able to get interrupts working.  I've tried two different "DORHEA 3PCS ESP32 Development Board" (Reviews on Amazon say it's really a  Doit V1).  I've also tried two different cc1101 boards.  The current one look like the one Aaron has.

I've triple checked the wiring and I think everything is correct.  All of the SPI signals look fine and the radio init routine is able to read the radio version (20).  Additionally cc1101_test_gpio() passes.  

As a test I put cc1101_test_gpio() in an infinite loop and watched GDO2 with a scope and I see a perfectly textbook square wave:

[image]

But when I run the normal code I get this funky waveform:

[image]

and zoomed out

[image]

Here's the log:

````
142 : SPI init
143 : SPI init done
143 : Radio init
154 : Radio version: 20
165 : Radio set to base freq: (F=4) 869.85 Mhz 
165 : Radio set id done 1
Error opening the settings file
265 : Interval timer init
265 : Interval timer done
*WM: [1] AutoConnect 
*WM: [2] [ESP32] event handler enabled 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: not_really_the_SSID
*WM: [3] Using Password: this_is_not_really_the_password
*WM: [3] WiFi_enableSTA enable
*WM: [3] Mode after delay:  STA
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.73.142
Connected! IP address: 192.168.73.142
mDNS responder started
*WM: [3] unloading 
3110 : Mode changed to Idle
3110 : Idle
3110 : Count: 0
3111 : Mode changed to Sync
3111 : Sync
Len 7 Data: 0x01 0xfe 0x02 0x00 0x00 0x00 0x00
3112 : Radio set to base freq: (F=0) 868.15 Mhz 
3565 : Count: 0
3565 : something wrong, back to idle
3566 : Mode changed to Idle
3566 : Idle
4665 : Count: 0
4665 : Mode changed to Sync
4665 : Sync
Len 7 Data: 0x01 0xfe 0x00 0x00 0x00 0x00 0x00
5765 : Count: 0
5765 : something wrong, back to idle
5765 : Mode changed to Idle
5766 : Idle
6865 : Count: 0
6865 : Mode changed to Sync
6865 : Sync
Len 7 Data: 0x01 0xfe 0x02 0x00 0x00 0x00 0x00
7965 : Count: 0
7965 : something wrong, back to idle
7965 : Mode changed to Idle
7966 : Idle
9065 : Count: 0
9065 : no interrupts anymore, something is broken, trying to fix it now
9065 : Mode changed to Sync
9066 : Sync
Len 7 Data: 0x01 0xfe 0x00 0x00 0x00 0x00 0x00
...
````

I'm out of ideas, anyone else have any ??

Thanks,
Skip

skiphansen, 2021-11-27:
Never mind, cockpit error!  I had commented out init_interrupt() while playing with the code before my cc1101 arrived!  DUH !
Interrupts work fine now.  (red face!)

closed 2021-11-27 by skiphansen

atc1441, 2021-11-27:
Good you found it out :)