atc1441/E-Paper_Pricetags #47: License missing

Hades32, 2021-12-29:
Hey  @atc1441,
would you mind adding an official `LICENSE` file? (GitHub will offer you templates for the most common ones)
I'd like to fork the access point for my use case (automatically pulling an image from the cloud) and would prefer to do that after the code is officially licensed.
It's gonna be open-source, too, so I don't mind which license you chose

Hades32, 2021-12-29:
That was fast! Thanks :)

closed 2021-12-29 by Hades32

atc1441, 2021-12-29:
Email push for the win :)

Hades32, 2021-12-29:
FYI, it will live [here](https://github.com/Hades32/web-tags).

I removed the need for git submodules and just kept the code of the access point. The latter will unfortunately mean, that rebasing changes will be manual work, but I didn't want to keep >100MB of files, just to keep the history...
If you ever consider splitting this in a "docs/experiments" and an "access point" repo, I would rebase my changes to make it easier to upstream later changes

atc1441, 2021-12-29:
Thanks for sharing your version as well.

As this whole "E-Paper Shelf Label" Topic is very versatile i did put everything in one repo to have everything from that topic in one place but i do understand what you mean.
In the next days i will also publish the Imagotag Access Point and it was also planed in a new folder but no i think about putting it into a seperate repo.

Im open to improvements

Hades32, 2021-12-29:
Yeah, multi-repo approaches are more work initially, and sometimes not worth when working alone on something, but when collaborating it's definitely the more flexible approach