atc1441/E-Paper_Pricetags #50: docs issue: What do Freq, NetID and Slots actually mean?

Hades32, 2022-01-16:
Although I could make everything work, I basically just used the default values because I didn't know what they mean (with the exception of the `Offset`, which *is* mentioned in the readme).

It would be nice if [the readme](https://github.com/atc1441/E-Paper_Pricetags/blob/main/Custom_PriceTag_AccesPoint/README.md) would mention:
* What is `Freq` (as it's 0-71 it doesn't seem to be an actual frequency)
* What is `NetID` (it sounds like subnets within the same frequency, but who knows. And can I use multiple `NetID`s at the same location? Potentially with a separate esp32)
* What is `Slots` (Does it limit the number of activated displays or the number of concurrent messages being sent?)

Bonus points for then adding a link to the `Settings` section of `index.html` that points to the readme, so people who normally don't read readmes (e.g. me) will find the information, too.

atc1441, 2022-01-16:
Will try to explain it here a bit just for basic info.

Freq is mapped to a defined frequency and after reversed values from the stock system. These are propriatary frequencys from the manufacturer.

NetId is a system id and a display will be assigned to one, no other system nearby should have the same one as you would not know on which it is synced to.

One slot happens every sync message, the number of slots defines how often a display wakes up in the end.

4 slots means the slot counter will reset each 4th sync and the display wakes up on its slot based on the set display id.

Slot
0 1 2 3 0 1 2 3 4 . . . And so on
On 8 slots:
0 1 2 3 4 5 6 7 0 1 2 3 4 5 6 7 . . . And so on

All these values will be set to a display on activation so a display needs to be reactivated if any of these values does change.

Hope that helps a bit. No bonus points right now for not adding it to the readme currently :)

Hades32, 2022-01-16:
I'll try to rephrase that in my own words, to see if I really understood that:
* `FreqID`: there's a list of 72 pre-defined frequencies by the manufacturer. This ID selects one of them
* `NetID`: just like a WiFi SSID. Must be different from your neighbor, but many displays that you own can have the same one.
* `Slot`: To save power, displays only listen to messages `100% / num_of_slots` of the time. So the more slots you define, the slower a device will react, but the more power it can save. The slot number it wakes up on, is calculated from the display ID (maybe something like `my_slot = my_id % num_of_slots`).

> All these values will be set to a display on activation so a display needs to be reactivated if any of these values does change.

That's definitely important info for the readme!

atc1441, 2022-01-16:
Yes that sounds like it is for me as well :)

This project was never really finished as there where some turbulences with the manufacturer which did stop me for that time and then another project came around ^^

These days i would do it like with the other system and use one display as AP and only connect uart to the esp32 so its a way more reliable connection

Hades32, 2022-01-16:
Cool, thanks. Added it to my docs: https://github.com/Hades32/web-tags/blob/main/docs/setup.md