atc1441/E-Paper_Pricetags #52: Custom_PriceTag_AccesPoint w/ 915MHz Chroma in US

combs, 2022-08-22:
Hello! I am excited by this project, it is very cool. Thank you for making it! 

I live in a region where 868MHz is not allowed but 915MHz is OK (US). So I acquired some Chroma74 (and other Chroma) units second-hand. They were sold in this region and were intended for installation in a store, but probably never installed. So they may be activated for 915MHz.

I see that there is a nice toggle in the Custom_PriceTag_AccesPoint UI to set the proper US wakeup frequency for CC1101. However, it seems like activation sets the activated units for 868MHz: https://github.com/atc1441/E-Paper_Pricetags/blob/main/Custom_PriceTag_AccesPoint/ESP32_Async_PlatformIO/RFV3/mode_activation.h#L167-L169

I can hack up my copy and get what I want for the activation process. However, I am having trouble raising any of the units with my CC1101/ESP32 combo. Has anyone successfully activated a 915MHz Chroma unit with Custom_PriceTag_AccesPoint? **Are there other CC1101 registers that need to be set for 915MHz access points?**

(I am using an 868MHz CC1101 green-board module from AliExpress. I have read it may perform OK at 915MHz once the proper registers are set. Within Custom_PriceTag_AccesPoint I know there is a frequency offset that needs to be set; I do not have a frequency analyzer or oscilloscope to determine the proper frequency offset, so am attempting to brute-force it by choosing every available frequency offset and trying a wake/full sync + activate. Checking for any response from units. So far nothing from -127 to 127. The radio initializes fine, as far as I can tell, and I think my wiring is good.)

Appreciate any help anyone can offer. Thank you!

combs, 2022-08-22:
starting to answer my own question, it looks like YES, the following registers are involved. I will read some datasheets until these registers make sense.

```

  spi_write_register(CC1101_REG_FSCTRL1, 0x06);
  spi_write_register(CC1101_REG_FSCTRL0, 0x00);
  spi_write_register(CC1101_REG_MDMCFG4, 0xCA);
  spi_write_register(CC1101_REG_MDMCFG3, 0x83);
  spi_write_register(CC1101_REG_MDMCFG2, 0x13);
  spi_write_register(CC1101_REG_MDMCFG1, 0x22);
  spi_write_register(CC1101_REG_MDMCFG0, 0xf8);
  spi_write_register(CC1101_REG_AGCCTRL2, 0x43);
  spi_write_register(CC1101_REG_AGCCTRL1, 0x48);// Not so hard on receiving data.
  spi_write_register(CC1101_REG_AGCCTRL0, 0x91);

```

clues from this thread:

https://e2e.ti.com/support/wireless-connectivity/sub-1-ghz-group/sub-1-ghz/f/sub-1-ghz-forum/533568/cc1101-works-at-868mhz-250kbps-msk-but-poor-performance-at-915mhz-msk-500kbps-using-a-specific-crystal-brand

```
This is the list of CC1101 registers that differs between 868MHz and 915MHz:
REGISTER 868MHz_250k 915MHz_500k
FSCTRL1: 0x0c 0x0e
MDMCFG4: 0x2d 0x0e
MDMCFG2: 0x71 0x73
MDMCFG1: 0xc3 0x42
AGCCTRL2: 0x07 0xc7
AGCCTRL1: 0x47 0x00
```

(although this is also changing from 250k to 500k and some other stuff, need to untangle)

combs, 2023-08-23:
ok, so I got SmartRF Tools set up and started playing around with register values. 

The changes from 868MHz to 915MHz are only in the `FREQ0`, `FREQ1`, `FREQ2` registers.

915MHz is defined by

```

    0x23,  // FREQ2         Frequency Control Word, High Byte
    0x31,  // FREQ1         Frequency Control Word, Middle Byte
    0x3B,  // FREQ0         Frequency Control Word, Low Byte

```

In the `Custom_PriceTagA_AccesPoint` application, these registers are set up through the "frequency selection" (0-71) configuration (confirm?) 

The table `cc1101_frequency_list` does include the 915MHz frequencies. So to set this frequency, it should only be necessary to put the right value in the "frequency" box in the webapp. I think it's index 35, zero-indexed. So I will enter this as a frequency and try all the offset values again for my module. wish me luck!

combs, 2023-08-23:
p.s., I got tired of scanning through frequency offsets manually, so here's a (not-particularly-pythonic) Python script to automate it.

https://gist.github.com/combs/23c81ebf9646539a667080a6ce3a9f74

combs, 2023-08-23:
`mode_activation.h` includes a hardcoded constant for EU, meaning that even if you choose a US frequency for the base station, it activates units to EU frequencies unless you change line 169 in `mode_activation.h`:

```

    case 13: //RF Region 0x40 = EU, 0x38 = US
      tx_act_buffer[8] = 9;
      tx_act_buffer[9] = 0x38;

```

amcolash, 2023-09-24:
Hey @combs, have you made any progress? I am trying your script and have changed the `mode_activation.h` file like mentioned. I don't have too much more patience with this display. I might just cut my losses and buy something that is designed to be hacked instead if I can't get it working. I didn't realize how difficult it would be to get this working!

combs, 2023-09-24:
> Hey @combs, have you made any progress? I am trying your script and have changed the `mode_activation.h` file like mentioned. I don't have too much more patience with this display. I might just cut [...]

Haven't gotten them talking yet. I just yesterday received a different EBYTE radio module to try, on the off chance that my "green module" from aliexpress wasn't actually equipped for 868/915MHz operation. Hope to try it out soon...

vonnieda, 2023-09-25:
Hi folks, I got this working previously but had set the project aside for some others. Here are some quick notes on how I got it working. Some of these notes are raw - forgive typos, etc.:

In platformio.ini change the lib_deps to this:

[image]

This should allow you to build E-Paper_Pricetags/Custom_PriceTag_AccesPoint/ESP32_Async_PlatformIO/platformio.ini with PlatformIO. I used the PlatformIO extension in VSCode. Any questions, ask. I don't know PlatformIO very well.

I used this ESP32 board: https://www.amazon.com/gp/product/B08PNWB81Z/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

With this RF board: https://www.amazon.com/gp/product/B01DS1WUEQ/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

And this antenna: https://www.ebay.com/itm/371850764045

(I had that antenna laying around for ADS-B, I didn't buy it specifically for this)

With these displays:
- https://www.ebay.com/itm/314849563630
- https://www.ebay.com/itm/155763019233

I used the US wakeup channel, which is 35. Calculated the frequency it should be from the code which was 914999969, or 915MHz. So then I started trying to activate them and watched the spectrum on HackRF, Windows, AirSpy. Could clearly see the wakeup calls but they were a little off 915. I played with the Offset till I landed on 35, which was right at 915. 

I also noted and converted the following frequencies from the source code:

```
0x21, 0x63, 0xF0, 868149902.34375 EU base
0x21, 0x66, 0xE4,
0x21, 0x6B, 0xD0,
0x21, 0x71, 0x7A, 869524963.37890625
0x21, 0x74, 0xAD, 869849884.033203125 EU wakeup
0x21, 0x33, 0x33,
0x21, 0x37, 0x23,
0x21, 0x3B, 0x13,
0x21, 0x3F, 0x03,
0x21, 0x42, 0xF4, 864799926.7578125
0x22, 0xB6, 0x27, 902499847.412109375 US base
0x22, 0xBB, 0x13,
0x22, 0xC0, 0x00,
0x22, 0xC4, 0xEC,
0x22, 0xC9, 0xD8,
0x22, 0xCE, 0xC4,
0x22, 0xD3, 0xB1,
0x22, 0xD8, 0x9D,
0x22, 0xDD, 0x89,
0x22, 0xE2, 0x76, 906999938.96484375
0x22, 0xE7, 0x62,
0x22, 0xEC, 0x4E,
0x22, 0xF1, 0x3B,
0x22, 0xF6, 0x27,
0x22, 0xFB, 0x13,
0x23, 0x00, 0x00,
0x23, 0x04, 0xEC,
0x23, 0x09, 0xD8,
0x23, 0x0E, 0xC4,
0x23, 0x13, 0xB1,
0x23, 0x18, 0x9D,
0x23, 0x1D, 0x89,
0x23, 0x22, 0x76,
0x23, 0x27, 0x62,
0x23, 0x2C, 0x4E,
0x23, 0x31, 0x3B, 914999969.482421875 US wakeup
0x23, 0x36, 0x27,
0x23, 0x3B, 0x13,
0x23, 0x40, 0x00,
0x23, 0x44, 0xEC,
0x23, 0x49, 0xD8,
0x23, 0x4E, 0xC4, 917999633.7890625
0x23, 0x53, 0xB1,
0x23, 0x58, 0x9D,
0x23, 0x5D, 0x89,
0x23, 0x62, 0x76,
0x23, 0x67, 0x62,
0x23, 0x6C, 0x4E,
0x23, 0x71, 0x3B,
0x23, 0x76, 0x27, 921999847.412109375
0x23, 0x7B, 0x13,
0x23, 0x80, 0x00,
0x23, 0x84, 0xEC,
0x23, 0x89, 0xD8,
0x23, 0x8E, 0xC4,
0x23, 0x93, 0xB1,
0x23, 0x98, 0x9D,
0x23, 0x9D, 0x89,
0x23, 0xA2, 0x76,
0x23, 0xA7, 0x62,
0x23, 0xAC, 0x4E,
0x21, 0x3C, 0x0F,
0x21, 0x3E, 0x07,
0x21, 0x40, 0x00,
0x21, 0x41, 0xF8,
0x21, 0x43, 0xF0,
0x21, 0x6A, 0x56,
0x21, 0x6C, 0x4E,
0x23, 0x5E, 0x07,
0x23, 0x5F, 0x04,
0x23, 0x60, 0x00,
0x23, 0x60, 0xFC,
0x23, 0x61, 0xF8, 919949951.171875
```

Bitmap resolutions of my displays:
- Chroma74 640x384
- Chroma29 128x296

Some other random notes:
- Aside from what I've mentioned here, I just followed the instructions at https://github.com/atc1441/E-Paper_Pricetags#stock-firmware-using-custom-ap
- Seems like maybe the 29s are EU Wakeup and 74s are US. I think I'm having better luck with that setup.
- Trying to send an image to the chroma74 reboots the ESP. In the compression, I think. Gotta see if I can pre-compress. I suspect the WROOM I was using was out of memory. Need to try with a WROVER that has SPI RAM.
- Found /draw.htm which is a nice little JavaScript draw thing, and it works great for the 29, but 74 still crashes. 
- If you reboot the ESP32 it can take like 15 minutes or more for everything to sync back up. It tripped me up a lot at the start because I was reflashing the firmware, and it seemed like displays would activate but then disappear. 
- I pulled the image conversion code out of the firmware and made a little command line thing in chroma_image_convert. I converted a JPEG I found on the weather thing page and uploaded it as a .bin and sent it and it worked!

[image]

[image]

[image]

combs, 2023-09-25:
oh wow, thank you very much, @vonnieda. I am grateful!