atc1441/E-Paper_Pricetags #59: Compilation ESP32_Async_PlatformIO in arduino

torx80, 2023-09-13:
Hi, I can't compile the RFV3.ino file from the "ESP32_Async_PlatformIO" folder via arduino. I also tried using Platfomio, but it throws up the message "UnknownPackageError: Could not find the package with 'me-no-dev/ESPAsyncWebServer' requirements for your system 'windows_amd64'", I use windows 10 x64

vonnieda, 2023-09-25:
I addressed this in https://github.com/atc1441/E-Paper_Pricetags/issues/52#issuecomment-1734502229. Please have a look.

torx80, 2023-09-26:
Through long-term research, I somehow miraculously managed to set Visual Studio to a state where it copies and uploads my project. I had to modify the platformio.ini file :

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
src_dir = RFV3

[common]
framework = arduino
lib_deps = 
	bblanchon/ArduinoJson@^6.17.3
	lib/ESP Async WebServer
    bitbank2/OneBitDisplay @ ^2.3.1

[base:esp8266]
monitor_speed = 500000
platform = espressif8266
framework = ${common.framework}
lib_deps = 
	${common.lib_deps}
	me-no-dev/ESPAsyncTCP

[base:esp32]
monitor_speed = 500000
platform = espressif32
framework = ${common.framework}
lib_deps = ${common.lib_deps}

[env:az-delivery-devkit-v4]
platform = espressif32
board = az-delivery-devkit-v4
extends = base:esp32

Next, I commented out the function: //mbedtls_md5_starts(&_ctx); in the library: WebAuthentication.cpp.

The system seems to work, I just don't have a way to test it because I don't have an unactivated ECL, the one I have is "borrowed" from some store. I don't know his ID. The only solution would probably be to download the FLASH content from the non-activated ECL and transfer it to the activated ECL. This would theoretically create identical copies suitable for activation. (reflection)   =)

torx80, 2023-10-06:
By the way, does anyone have FLASH content
 from non-activated CC1110 from Chroma 74 BWR?